tanelih/phonegap-bluetooth-plugin #30: bt socket connect failed, code:9

CosminPOP, 2015-04-14:
Hi

I'm getting this error whenever I try to 'connect' to any device:
"bt socket connect failed" error code: 9

I added an 'error' parameter to the __onFail__ function :

...
var onFail = function (*error*) {
...
 
to see this error response that gets called from

...
window.bluetooth.connect(onConnection, __onFail__, {
                uuid: device.uuids[0],
                address: self.model.get('address')
            });
...

tanelih, 2015-04-16:
Have you enabled Bluetooth on both devices? Are you sure the device address is set correctly?

There could be many reasons for this error, the code you are receiving (`9`), refers to [this error](https://github.com/tanelih/phonegap-bluetooth-plugin/blob/master/src/android/BluetoothError.java#L20). And it seems like the error happens somewhere around [here](https://github.com/tanelih/phonegap-bluetooth-plugin/blob/master/src/android/BluetoothWrapper.java#L825-L854). 

Try adding some logging somewhere in the area I pointed out in order to pinpoint the actual error, or failure that is happening. Good luck!

MalikDrako, 2015-04-21:
I'm getting this same error.
I can get UUIDs and pair the devices, but when I try to connect with any of the retrieved UUIDs I get:

    Connect error:  Object {message: "read failed, socket might closed or timeout, read ret: -1", code: 9}

I am running the app on a Nexus 9, trying to connect to a Nexus 7 and a Windows 8 PC. I can connect to the PC through the bluetooth settings, it appears on the PC as a stereo audio service. 
The example app does the same as my app, although I had to add extra logging in the JS to verify this.