tanelih/phonegap-bluetooth-plugin #31: Discovery and Pairing

splinter7, 2015-06-16:
Hi,

I need to be able to create an app that will discover a Bluetooth device and pair it with a mobile device (smartphone/tablet). I came across this Cordova plugin as I am using Cordova to build the application. I have installed the plugin and build the apk from the example included. The enable and disable buttons work as expected but the discovery button doesn't display any results even though I have other Bluetooth devices on and ready for discovery.

The second part of my question is after device discovery how would I pair?

Thanks and Regards

tanelih, 2015-06-16:
Hey,

What version of Cordova are you using? Back when I was (relatively) active in developing the plugin I think the Cordova version was something like 3.0, so I can't really say everything works fine if you're using the latest versions. Take a look at what the application is logging when you try to discover devices, the `console.log` commands should be sent to the Eclipse debugger IIRC.

I think you might want to take a look at some of the more recently updated plugins, that also support the low energy stuff.

splinter7, 2015-06-16:
Hello Taneli,

I'm using the latest version of Cordova 4.0. I'm not currently using Eclipse as my dev environment I'll probably have to set that up. I tried looking at a couple of the newer plugins but yours is the only one that I have working in any form so far. 

I'm assuming my test devices may lack proper support for BLE just guessing here. I'm using a Nexus 7 2013 and a Sony Xperia M (first generation). I'm trying to pair them with a Bluetooth 3.0 headset. 

Thanks for your response.

tanelih, 2015-06-16:
Hi,

I have no testing experience with Cordova 4., so I can't really guarantee this plugin works properly with that. I've not tested this plugin with any BLE devices so I can't really comment on that.

Hope you get your app working!

splinter7, 2015-06-16:
Thanks Taneli,

I understand. Not sure what my next move is yet. Thinking I may just have
to exclude Bluetooth integration for now if I can't find a working solution
in the next few days. Thank you for your help.

Regards,
Stewart

splinter7, 2015-06-18:
Hi Taneli,

I continued to play around with your plugin I have found that the discovery works. It just doesn't display the list of devices found after discovery is run. I'm not familiar with backbone.js or underscore.js so it's not clear to me how to make this happen. Do you think you can assist me here.

Regards,
Stewart

tanelih, 2015-06-19:
Hey,

Glad to hear you got the discovery working.

Do you have a way to see the output of the `console.log` commands from the JavaScript code? You should be able to verify that you are discovering the devices by listening to the `console.log` [here](https://github.com/tanelih/phonegap-bluetooth-plugin/blob/master/example/www/main.js#L218). If you're seeing any errors or warnings from the code, that might also help finding the solution to this problem.

With backbone.js the "flow" of the device data from the actual plugin code to the front end you're seeing (or currently not seeing), is relatively simple. Discovered devices are wrapped in a `Device::Backbone.Model` instance and added to the single `DeviceCollection::Backbone.Collection` in the example application. This collection is set as the "data source" for the `DeviceListView::Backbone.View`, which listens to the `reset` and `add` events on the collection and renders itself. 

Check out [this](https://github.com/tanelih/phonegap-bluetooth-plugin/blob/master/example/www/main.js#L211-L236) code to see the device discovery process which takes the device data received from the plugin and adds it to the backbone collection.

It might also be useful to check that the `BluetoothState` is not jammed in the `Busy` state for some reason.

There is also the scary fact that the packages mentioned in the `bower.json` file are not locked to the specific version but are relaxed in terms of `patch` updates. These updates should never break any backwards compatibility but you never know. Check which versions of the dependencies you are running,  and test if you can get the specific versions mentioned in the `bower.json` by removing the `~` character from before the semver number.

splinter7, 2015-06-19:
Hello,

I'm using alerts instead of the console.log I can see from my alerts that my device is discovered I had to modify your code a little but it returns the device address and name.

I am reading through your code and I'm beginning to slowly decipher it :)

I have the latest versions of the packages from the bower.json file but his is because I downloaded and added them myself. I revert to the versions in the bower.json and let you know if anything changes.

Thanks again

splinter7, 2015-06-21:
Hey,

I have the code working now the way I want it. I am pairing my android phone with a headset. Everything works as expected and the phone is paired. My concern now is the headset it's actually playing any audio. When I pair the devices through the phone's Bluetooth settings the audio is played through the headset.

Any ideas why I might be having this issue

Thanks

tanelih, 2015-06-21:
Hi,

I'm not sure if it's possible to transmit audio using the basic `BluetoothDevice` instance used by the plugin. I think you might need to edit the plugin code to use the `BluetoothHeadset` if the `BluetoothClass` of the device happens to be something akin to "headset". You might need to modify the plugin code so that it sets the paired device as the audio output if it's something like a headset or some sort of an audio output capable device.

Take a look at the [documentation](http://developer.android.com/reference/android/bluetooth/BluetoothHeadset.html) for `BluetoothHeadset` and [Bluetooth](http://developer.android.com/guide/topics/connectivity/bluetooth.html) in general. I've no experience with audio and Bluetooth, so I'm afraid I can't really offer much help with that matter.

If you need to modify the plugin code to support audio devices, I suggest creating a fork of this repository and then adding your modified fork as a dependency to your project.

gcdpinho, 2016-04-02:
Hi, forgive me for my english, i'm brazilian.
I read the previous posts and have a similar problem. The discovery works fine, but my alert in the function onDeviceDiscovered show me Object Object, instead of the device.

onDeviceDiscovered = function(device) {
			alert('found device ' + device);
			devices.collection.add(new Device(device));
		}

Thanks, and sorry abount my english again.

tanelih, 2016-04-02:
Hey,

The alert coerces the device object into a string, hence the `[Object object]` notation. You'll probably want to use the following way to verbosely log out the object you receive from the callback:
```javascript
alert('found device ' + JSON.stringify(device) + '');
```

gcdpinho, 2016-04-06:
My problem, besides this, was with versions of packages. Now, it's ok, but when i click on the button connect, next to device, it always enters onFail, when this function is called:

window.bluetooth.connect(onConnection, onFail, {
				uuid:    device.uuids[0],
				address: self.model.get('address')
			});

I know it's hard to know what it is, having a superficial view of the problem, but can you help me?

tanelih, 2016-04-06:
Can you print out the error that you're receiving? That would be a start. 🕵

gcdpinho, 2016-04-06:
[image]

this alert it refers to that function: 
`var onFail = function(error) {`
			`alert(JSON.stringify(error));`
			`Bluetooth.set({`
				`state: BluetoothState.Ready`
			`});`
			`self.$('.btn-bt-connect').button('reset');`
		`}`

tanelih, 2016-04-06:
What is the device you're trying to connect to? Are you sure there is a bluetooth server accepting connections in it?

gcdpinho, 2016-04-07:
I managed to solve the problem: on some devices i had to catch device.uuids[1], because the uuids[0] was giving error(I don't know the reason). Now, i need change message between two android devices, and the function write is working. I saw the parameter onDataRead in startConnectionManager, but I don't know how to read the message received.