tanelih/phonegap-bluetooth-plugin #32: Print

danicarla, 2015-08-06:
Hi
How i can use this plugin to print a ticket?

tanelih, 2015-08-11:
Hey,

You should be able to `connect` and `write` to and `read` from any generic bluetooth devices (such as printers). Check out the [example](/tanelih/phonegap-bluetooth-plugin/tree/master/example) for how to establish connections and manage them.

You might also want to take a look at #13 since it contains a reasonable example on how to send data to devices.

danicarla, 2015-08-18:
Excuse
The example did not me help

I could connect... but can not do anything else

tanelih, 2015-08-18:
Hey,

Check out #13 for an example of how to write data to the connected device.

danicarla, 2015-09-03:
Sorry.. but dont worked :(
I am using Intel XDK, That would be the problem ?

tanelih, 2015-09-04:
I can't really say. I've no experience with it.

danicarla, 2015-09-15:
Hello. Sorry it took me to answer.
The print worked with the code you provided me!!!

```
<script>
    var onSuccess = function() {
      console.log('Wrote data succesfully.')
    }
    var onFailure = function(err) {
      console.log('Something went wrong...', err);
    }          
</script>    
<a href="#"  onclick="bluetooth.write(onSuccess, onFailure, 'Hi Dani, How are You?')">Click to Print</a>
```
But I have to click three or four times to print...
I think my mistake is with Javascript :(

tanelih, 2015-09-16:
Hi,

To make sure the `onclick` handler is actually firing, you can do something like this instead:
```html
<script>
  var onSuccess = function() {
    console.log('Wrote data succesfully.');
  }
  var onFailure = function(err) {
    console.log('Something went wrong...', err);
  }          
  var onClick = function onClick() {
    console.log('I have been clicked!');
    bluetooth.write(onSuccess, onFailure, 'Hi Dani, How are You?');
  }
</script>
<a href="#" onclick="onClick()">Click to Print</a>
```

Also take note that since the operation is asynchronous it will take some time to complete, so you shouldn't necessarily see the results instantly. Although usually they are pretty instant I guess...

danicarla, 2015-09-22:
Hi tanelih
The log not show anything unusual.. 
see the operation in video: https://www.youtube.com/watch?v=pFLX4xi-Z6s

And see the Log:
```
>bluetooth enabled
 state changed 3

>starting discovery...
 state changed 2

>found device 
 Object

    address: "00:08:1B:95:3A:CA"
    name: "P25_027069_01"
    __proto__: —

>Connect 

>I have been clicked!
 Wrote data succesfully.

>I have been clicked!
 Wrote data succesfully.

>I have been clicked!
 Wrote data succesfully.

>I have been clicked!
 Wrote data succesfully.

>I have been clicked!
 Wrote data succesfully.

>I have been clicked!
 Wrote data succesfully.
```
Thank's for your patience

tanelih, 2015-09-23:
Hi,

Cool video, it really helps to see the system in action. It looks like the message is printed as many times as you've clicked the write button. So in a sense the plugin seems to work just fine?

Have you tried writing a longer message, which would force the printer to write multiple lines with a single write command. There might be some sort of a minimum for the message length before it starts printing each message.

Is there a specific "code" to tell the printer to print (something like EOF)? You might need to write in something like that in order for the printer to recognize that it needs to stop waiting for input and print.

danicarla, 2015-09-23:
> What happens if you click the write once and then press the button in printer to spit out the paper. Has it written the message still?
> -  tanelih

If clicked only one time, and press the button in printer to spit out the paper comes out blank 

> Have you tried writing a longer message, which would force the printer to write multiple lines with a single write command. There might be some sort of a minimum for the message length before it starts printing each message.
> -  tanelih

I do not think of it before. I realized a new test with a long message and it worked !!! :blush: :smile: :relaxed:

Not print few characters that are at the end of the long message. 
In the first test were the last 37 characters, And In the second test with bigger message 46 the last characters.

Apparently he has a minimum length for the start and the end of the information sent.

> Is there a specific "code" to tell the printer to print (something like EOF)? You might need to write in something like that in order for the printer to recognize that it needs to stop waiting for input and print.
> -  tanelih

I will search these codes. because I want to print images too. 

Thanks for your help , I try finish my project.. And if successful, i'm back to feedback

tanelih, 2015-09-24:
Good to hear you got it working. 

In the original use case for which the plugin was developed for, we had to write ACK messages to the device to let it know we had the data and that it could send more. Maybe the printer expects similar messages?

danicarla, 2015-09-24:
Cool ... I'm enjoying working with it
There's a repository? where I can see those messages that you have written ?

tanelih, 2015-09-24:
Unfortunately I don't know where the repository is anymore. So I can't really help with that.

danicarla, 2015-09-24:
No problem!
Helped me a lot!
I'll keep trying

Thanks :wink:

danicarla, 2015-10-01:
Tanelih
Could you help me to send text messages in bold to printer?