tanelih/phonegap-bluetooth-plugin #35: getUuids fetching lasts too long

bandeye, 2015-09-09:
Hi Tanelih!

I'm trying to get the UUIDs of the given mac address, but when I use the function below, it lasts too long, and I'm wondering what am I doing wrong.

window.bluetooth.getUuids(onSuccess, onFail, mac_address);

All I can see in logcat is posted below:

"
VFY: unable to resolve direct method 1783: Ljava/lang/String;.<init> ([BLjava/nio/charset/Charset;)V
VFY: unable to resolve virtual method 99: Landroid/bluetooth/BluetoothSocket;.isConnected ()Z
VFY: unable to resolve virtual method 99: Landroid/bluetooth/BluetoothSocket;.isConnected ()Z
"

And if I want to call the function again, it says, totally logically:

"
Could not start UUID fetching because there is already one in progress.
"

Snippet:

function getUuidForDevice() {
	window.bluetooth = cordova.require("cordova/plugin/bluetooth");
        var mac_address = AA:BB:CC:DD:EE:FF // just an example
	window.bluetooth.getUuids(onSuccess, onFail, mac_address);
}

function onSuccess(device) {
	alert("Name of the device: " + device.name);
	alert("Address of the device: " + device.address);
	for (var i = 0; i < device.uuids.length; i++) {
		alert("UUid" + i + " of the device: " + device.uuids[i]);
	}
}

Can you help me to figure out, what is wrong?

Thanks in advance,
bandeye

tanelih, 2015-09-09:
Hi,

After googling around, it seems that this problem could be caused by having insufficient permissions in the manifest file, or having an old version of android.

Some resources also pointed to android studio being the culprit here. I'd try to clean and rebuild the project and making sure you're running a high enough version of the android sdk.

The code you posted seems to be in order. Are you sure you've "discovered" the device so that your device knows about it?

bandeye, 2015-09-09:
Hi Tanelih,

Thank you for getting back to me. Yes, you are right, the problem was the API version. The device supported API 8, which, as turned out, is the problem. I tried with another device, having API version 16, and it works. Thank you.

closed 2015-09-09 by bandeye