tanelih/phonegap-bluetooth-plugin #36: Issue in the bluetooth connectivity

goldyhmr237, 2015-09-28:
Hi,
i have issues regarding your code. when i am connecting device with bluetooth then it jumpsout from the function ( var gotUuids = function(device)  ) after alert and disconnecting the device.
please provide me a solution regarding this issue.

Thanks

tanelih, 2015-09-29:
Can you provide the code you are experiencing problems with? You can three backticks (```) to paste code.

goldyhmr237, 2015-09-29:
``` 
connect: function() {
	var self = this;
	Bluetooth.set({
		state: BluetoothState.Busy
	});
	self.$('.btn-bt-connect').button('loading');
	var onFail = function() {
		Bluetooth.set({
			state: BluetoothState.Ready
		});
		self.$('.btn-bt-connect').button('reset');
	}
	var gotUuids = function(device) {
		alert('got UUID\'s for device: '+ device.uuids[0]);
		var onConnection = function() {			
			self.model.set({
				isConnected: true
			});
       		alert('got connection');
			var onConnectionLost = function() {
				alert('lost connection');
				self.model.set({
					isConnected: false
				});
				onFail();
			}
			alert('output data to console...');
			window.bluetooth.startConnectionManager(
				console.log, onConnectionLost);
		}
		window.bluetooth.connect(onConnection, onFail, {
			uuid:    device.uuids[0],
			address: self.model.get('address')
		});
	}
	window.bluetooth.getUuids(gotUuids, onFail, self.model.get('address'));
},
```
this is my code, it's alerting got UUID\'s for device: '+ device.uuids[0] but it doesn't work or jumpsout from onConnection function because it's not alerting ('got connection'); and Here is my .apk file http://softweavertech.com/projects/android-debug-unaligned.apk . Please check it and let me know if any wrong with my code.

Currently it is finding Bluetooth devices near it, but when i click on Connect button, nothing happens.

Thanks

tanelih, 2015-09-29:
Is the `window.bluetooth.connect` function triggering the `onFail` callback? If it is, you can get the error as the first parameter to the `onFail` function.

goldyhmr237, 2015-09-30:
Yes, ``` window.bluetooth.connect ``` function is trigerring on the onFail callback. What should I alert in onFail function, so I see the error ?

tanelih, 2015-10-01:
Hi,

Sorry for the late reply. You can do something akin to:
```javascript
var onFail = function(err) {
  alert('ERROR:: ' + err); 
  Bluetooth.set({
    state: BluetoothState.Ready
  });
  self.$('.btn-bt-connect').button('reset');
}
```

goldyhmr237, 2015-10-01:
``` 
var onFail = function(err) {			
    var out = '';
	for (var p in err) {
        out += p + ': ' + err[p] + '\n';
	}
	alert(out);
    //alert('ERROR:: ' + err); 
	Bluetooth.set({
	    state: BluetoothState.Ready
    });
    self.$('.btn-bt-connect').button('reset');
} 
```

Hi, 
Here is the code which gives me an error, and I also sending you screenshot what is the error please check the attachment.

[image]

Thanks.

goldyhmr237, 2015-10-01:
I have enabled bluetooth on both devices. How to ensure that the device address is set correctly?

tanelih, 2015-10-01:
Hey,

I'm not sure about your use case. I haven't tested connecting to other mobile devices. 

Maybe the device in question is not accepting incoming connections? At least the error you're getting seems to indicate that.

goldyhmr237, 2015-10-01:
Hi,
I using app on HM NOTE 1LTE (XIAOMI) phone and connection two other devices which are Nokia 
5233 and Micromax bolt AD3520.

Thanks.

tanelih, 2015-10-01:
You might want to check that you're running some sort of an app on the Nokia and Micromax devices, which can accept incoming connections. It's not necessarily enough that they're Bluetooth enabled, they need to also be able to accept incoming connections.

tanelih, 2015-10-01:
Check #30. You're basically experiencing the same issue as the folks over there. You need to have some sort of a "slave" device, which can accept incoming connections. For example in the issue I linked the connection was succesful when the receiving end was some sort of a Bluetooth speaker, a "slave" in a sense.

Nullpo, 2015-11-25:
Hi!

With this plugin, Can I put an android device in that "slave" mode?

tnx!

tanelih, 2015-11-26:
I don't think so, sorry. You can only connect to devices, not receive connections from.

Nullpo, 2015-11-26:
No problem. Thank you!!!!! :D