tanelih/phonegap-bluetooth-plugin #38: Android 6 (Marshmallow) support

estanall, 2015-12-17:
First off, thanks for the great plugin.

Second, Android 6 has added a few new Bluetooth permissions (see http://developer.radiusnetworks.com/2015/09/29/is-your-beacon-app-ready-for-android-6.html).

I think that the only change that needs to be made is to add this line:
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
to plugin.xml. I did that in an app I wrote and it worked in Android 4.2.2 and Android 6.0.1. If this permission is not included, then newer Android devices can start a Bluetooth discovery but they cannot receive device information when they are found.

tanelih, 2015-12-22:
Hey,

Good catch, but does it break compatibility with older versions of android?

If you could make a pull request that would be great. Sorry for the late reply I'm swamped with work. 🐜

tanelih, 2015-12-22:
Ah sorry I didn't notice you mentioned it works with older versions as well. 🌚

closed 2015-12-22 by estanall