tanelih/phonegap-bluetooth-plugin #40: How to send ESC Commands?

danicarla, 2016-03-28:
Hi.... 

I know that my question is not a phonegap-bluetooth-plugin
But I believe you can help me .

I would like to know how to send ESC Commands to a printer
Any feedback is appreciated!

Tks !

tanelih, 2016-04-02:
Hi, 

I've no idea. Sorry. 😓

danicarla, 2016-04-05:
I appreciate the help! :)

The support my printer said I'm need to send the command as HEX string, not plain text.
But I have no idea how to do it

tanelih, 2016-04-05:
You'll need to convert the `ESC` string to a hexadecimal representation. You can try an online tool like [this](http://string-functions.com/string-hex.aspx) or just do the conversion in your code.

danicarla, 2016-04-08:
Great!!! I think I'm evolving.

I had already tried that way, but even so it prints the hex code... 
I'm send the HEX code in an application developed by the company that distributes the printer and worked correctly ...

But in my application does not work :(

I 'm thinking I need to add some code at the beginning , to inform you that will be sent "HEX string" instead of "Plain Text"

dannygunawan, 2016-04-17:
hi @danicarla 
have you found any solution for this issue?

danicarla, 2016-05-09:
@dannygunawan 
Sorry.. Unfortunately you still can not print

dannygunawan, 2016-05-11:
@danicarla Just in case you still need this feature, I found another plugin as an alternative. [https://github.com/don/BluetoothSerial]

danicarla, 2016-05-12:
@dannygunawan Great!!!
When I have time.. I will test
Tks!