tanelih/phonegap-bluetooth-plugin #42: compatibility with Motorola CS3070

larkonesvk, 2016-06-27:
Hello, 
every time i want to connect to barcode scanner a got: READ FAILED, SOCKET MIGHT CLOSED OR TIMEOUT, ERROR CODE: 9

my code is:
`onDeviceReady: function () {
    var dev = {};
    app.receivedEvent('deviceready');
    window.bluetooth.enable(function () {
        alert("SUCCESS");
    }, function () {
        alert("FAILED");
    });
    window.bluetooth.startDiscovery(onDeviceDiscovered, onDiscoveryFinished, onError);

    function onDeviceDiscovered(device) {
        alert("Found device " + device.address);
        if (device.address === "40:83:DE:4B:D4:12") { // address of scanner
            window.bluetooth.getUuids(onUuidsRetrieved, onError, device.address);
        }
    }
    function onDiscoveryFinished() {
        alert("SUCC FINISHED");
        if (dev.address === "40:83:DE:4B:D4:12") {
            alert("Trying to connect...");
            window.bluetooth.connect(onConnected, onErrorConn, {
                address: dev.address,
                uuid: dev.uuids[0]
            });
        }
    }
      function onConnected() {
        alert("Connected");
    }

    function onUuidsRetrieved(device) {
        alert(device.address + " UUID: " + device.uuids);
        if (device.address === "40:83:DE:4B:D4:12") {
            dev = device;
        }
    }

    function onError() {
        alert("ERROR");
    }

    function onErrorConn(error) {
        alert(error.code + " " + error.message);
    }

}`

is there something i missing? 

thanks

tanelih, 2016-06-28:
Is the barcode scanner in a mode where it's accepting connections? Have you paired the device with it? Are you sure your `dev.address` is correct? Unfortunately there can be a multiple reasons for this error... The code seems fine though, so there's that. :-)

larkonesvk, 2016-06-28:
Thanks for your reply.
I forgot to write that **yes** devices are paired and **yes** address is correct. I didnt try only this plugin but i also tried many others and got the exactly same error. I just start thinking that its impossible to connect from any cordova bluetooth plugin to bluetooth barcode scanner. Barcode scanner is motorola cs3070.

tanelih, 2016-06-28:
Hi,

I checked the specs of the CS3070 and it seems as though it features connectivity with iOS & Android... Most of the cordova bluetooth plugins likely use the native Bluetooth API's of the platform, similarly to this plugin. 

You're "discovering" the device though? Does the device have some sort of a "mode" that needs to be on before it's connected to. 

Do note that this plugin is essentially a Bluetooth client and can't act as a server. However in this case I'm guessing the scanner should be the "server".

larkonesvk, 2016-06-28:
`Do note that this plugin is essentially a Bluetooth client and can't act as a server. However in this case I'm guessing the scanner should be the "server".`

My point is to create cordova app for android which can discover bluetooth device (and so my scanner) and established connection between them. And then i want to send scanned data via bluetooth to my app. So i think my device should be server and scanner client. And as you said this plugin can't act as server my work ends here i think.

tanelih, 2016-06-28:
I mean "server" as in the device which accepts connections and doesn't initiate them. If your app is initiating the connections then it's essentially acting as a "client" in this case. 🌚

larkonesvk, 2016-06-28:
Thanks for your time and quick replies.

closed 2016-06-28 by larkonesvk